stm32duino/STM32duinoBLE #30: BLE transfer speed on STM32WB5 and corrupted data

Kamik18, 2021-05-19:
I am limited to 150-180 bytes/s on the transfer speed when I write data from the STM32WB55 to a RPI with read and notify. 
I have tested different timings for when the device poll and the advertising interval. Furthermore, I have tested with different buffer sizes between 8 and 22, and the data is corrupted if try to send more data than 22 bytes.

Result from a 30 byte uint8_t buffer of zeros: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01.

Is this a config error? and can the speed be increased?

Best regards Kasper G. Mikkelsen

fpistm, 2021-05-26:
Hi @Kamik18 
Please, could you share you test sketch?

What is your config? Arduino IDE version?  Core version? Library version? Selected options?...
all relevant information are welcome to help.

Kamik18, 2021-05-26:
I am using PlatformIo as my Arduino IDE, and I have tried changing BLE_DEFAULT_ATT_MTU but without any influence on the data.

I tried modifying one of your examples (CallbackLED.ino, examples/Peripheral/CallbackLED/CallbackLED.ino), and kept every variable to their default values, where I used “BLENotify” to write a data buffer in a while loop. I did not use delay or anything else doing that test.
I furthermore tried creating a callback function to send data when requested by a RPI, and again without any delay on either the STM32 or the RPI.  

Versions:
PlatformIo Core 5.1.1 and Home 3.3.4
Arduino_Core_STM32 2.0.0
STM32duinoBLE 1.2.0

KMeldgaard, 2021-06-04:
I've tried using [GrumpyOldPizza's ArduinoCore-stm32wb implementation](https://github.com/GrumpyOldPizza/ArduinoCore-stm32wb) on the Arduino IDE (following the README guide) and achieved impressive bandwidth. I can transfer ~120 bytes per package - speed is yet to be determined, but it is definitely faster.
However I still have issues  migrating the project to PlatformIO.

fpistm, 2023-01-19:
I've tested and read and write up to 256 and have met no issue with latest core 2.4.0, latest library version 1.2.2 and latest wireless copro binary.