stm32duino/STM32duinoBLE #34: Configurable BLE application

KMeldgaard, 2021-11-11:
Dear STM32Duino team

I want to change the BLE app configurations found in [app_conf.h](https://github.com/stm32duino/STM32duinoBLE/blob/main/src/utility/STM32Cube_FW/app_conf.h) (generated with STM32CubeMX I presume).
The only way possible I can see at the moment, is forking this repo, risking to fall behind updated versions.
Is it in the pipeline to make the BLE interface configurable?

Regards
K_Meld

fpistm, 2021-11-12:
Hi @KMeldgaard 

you're right.
One way should be to rename `app_conf.h` to `app_conf_default.h`
Don't forget to protect all define which should/could be redefined with `#ifndef`/`#endif`. ex: 
```c
#ifndef CFG_BLE_HSE_STARTUP_TIME
#define CFG_BLE_HSE_STARTUP_TIME  0x148
#endif
```

Then create a `app_conf.h` with:

```c
#ifndef APP_CONF_H
#define APP_CONF_H

#if __has_include("app_conf_custom.h")
#include "app_conf_custom.h"
#endif
#include "app_conf_default.h"

#endif /* APP_CONF_H */
```

KMeldgaard, 2021-11-15:
Hi @fpistm 

Thanks for the quick response.
I've tested out your solution in my fork, and so far its works like a charm.
Do you accept a Pull-request to the official repo?

Regards

fpistm, 2021-11-15:
Of cousre :)
All contributions are welcome.

closed 2021-11-15 by fpistm