stm32duino/STM32duinoBLE #41: Receipt by polling ?

YannickMarietti, 2022-03-28:
Hello

This is not really an issue but rather a comment on a possible improvement.
From what I understand, in current implementation, STM32duinoBLE is managing events via polling :

 __// poll for BLE events
  BLE.poll();_

This can be an issue because you can experience delays (or fails ?) on BLE communication if the loop() function is managing a time-consuming code.

Is it possible to manage them by interrupts, for example triggering an IT attached to a given characteristic when a central take an action on it ?

Best regards

Yannick

fpistm, 2022-03-28:
Hi @YannickMarietti 

In fact this library is based on official ArduinoBLE:
https://github.com/arduino-libraries/ArduinoBLE

So we follow the design and API of the official one:
https://www.arduino.cc/en/Reference/ArduinoBLEBLEpoll

You should submit this request to the original library then we could inherit.

The only event are for connected/disconnected: https://www.arduino.cc/en/Reference/ArduinoBLEBLEsetEventHandler

YannickMarietti, 2022-03-28:
Thanks !
Good to know :)
I wonder why they didn't permits to manage other events.
So, I'm transferring the request on https://github.com/stm32duino/SPBTLE-RF, is that right ?

fpistm, 2022-03-28:
> Thanks ! Good to know :) I wonder why they didn't permits to manage other events. So, I'm transferring the request on https://github.com/stm32duino/SPBTLE-RF, is that right ?

Welcome. 
No to this one: https://github.com/arduino-libraries/ArduinoBLE 😉

closed 2022-03-28 by fpistm