stm32duino/STM32duinoBLE #53: can't get P-nucleo-WB55RG (MB1355D) to work

thijses, 2023-03-21:
Dear creator(s) of STM32duinoBLE,
This library would be an awesome tool for a project i'm doing, BUT,
i can't get past BLE.begin() in the example sketch.
I did some minor digging (with the intent to save you some time) using the debugger, and it makes it to here:
BLE.begin()
  HCI.begin()
    _HCITransport->begin()

fpistm, 2023-03-21:
Hi @thijses 
to use the WB55RG you have to update the copro binary as stated in the README.md:

> In order to use this library with SM32WBxx series, you need to update the STM32WB Copro Wireless Binaries
> with stm32wbxx_BLE_HCILayer_fw.bin depending of your mcu:
> https://github.com/STMicroelectronics/STM32CubeWB/tree/master/Projects/STM32WB_Copro_Wireless_Binaries
> Each subdirectories contains binaries and Release_Notes.html which explain how to update it.

A new STM32WB has been released and I do not yet test it. So I advice to use the v1.15.0:
https://github.com/STMicroelectronics/STM32CubeWB/tree/v1.15.0/Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x

fpistm, 2023-03-21:
IIRW, yes. I noticed sometimes Cube programmer requires to hit twice, for example I always hit twice Start FUS. Problem is raised internally and hope will be fixed, but not on my scope.

thijses, 2023-03-21:
alright, dang. I feel a little silly, but my problem is solved!
thank you for your incredibly speedy reply

for future TL;DR:
i just needed to try pressing "start Wireless Stack" in STM32CubeProgrammer after uploading
i just tried with FW 1.16.0 and it worked, but only on the second time i used the start button for some reason

closed 2023-03-21 by thijses