stm32duino/STM32duinoBLE #64: Add support for Nucleo-F446RE

Cbannon35, 2023-12-08:
It is my first time doing any sort of microcontroller development, so please forgive me if I'm making silly mistakes.

I am trying to use the arduino IDE to use BLE on my Nucelo-F446RE board. Honestly don't have a whole lot of context for this issue, but in the `LED.ino` example:

```C
void setup() {
  Serial.begin(115200);
  Serial.println("starting");
  while (!Serial)
  // set LED pin to output mode
  pinMode(ledPin, OUTPUT);
  // begin initialization
  if (!BLE.begin()) {
    Serial.println("starting BLE failed!"); // never reached

    while (1);
  }
  Serial.println("starting BLE!"); // never reached
```

Neither print statement is reached, leaving me to believe that my system is hanging in the call to `BLE.begin()`. I tried debugging and that seems to confirm my suspicions (??). 

As a complete novice to stm32 boards / microcontroller development I would love for things to just work 'out of the box' and I'm wondering if this is an issue where my board is not supported by this library.

fpistm, 2023-12-09:
Hi @Cbannon35 
Nucleo F446 has no BLE module. Do you have one? Which one?

Cbannon35, 2023-12-09:
LMAOOOO you're kidding. No I totally thought it was built in... do you have a recommendation for a ble board that draws minimal current?

fpistm, 2023-12-09:
Nucleo WL55JC. Or if you kept Nucleo F446 see the README.md for the supported BLE shield.

closed 2023-12-09 by fpistm