stm32duino/STM32duinoBLE #65: mtu size unclear

thijses, 2024-01-04:
I'm trying to determine what the maximum package size for my STM32WB55 based application should be.
- CFG_BLE_MAX_ATT_MTU defaults to 156
  (but SHCI_C2_BLE_INIT_OPTIONS_LL_ONLY is enabled, so this can be overridden???)
- using an ST-link debugging session, ATTClass._maxMtu (and the _peers[0].mtu) read as 242 (which is readLeBufferSize() pktLen (251) - 9. Presumably determined by CPU2 in some way, right?)
- anecdotally the largest packet that the MCU acknowledges is 115 bytes (both from ST BLE Toolbox app and NFR Connect app)
so which one is it? (and which macro/function can provide the correct number in the future?)

P.S.
i am trying to make a basic OTA application, but the ST BLE Toolbox app insists on using 244 byte packets. I know that BLE 5.0 can technically handle an MTU of 512(ish?), but i'd be willing to wait a little longer and just use the 20-byte packet structure of BLE 4.0 if that were an option.

fpistm, 2024-01-11:
Hi @thijses 
It is not possible to change it as stated here:
https://github.com/stm32duino/STM32duinoBLE/blob/1c3d6070189e813b4edbc4aaf3aa52717e06c9cf/src/utility/STM32_WPAN/app_conf_default.h#L177C16-L177C16

The MTU size is defined by the CPU2.

thijses, 2024-01-13:
Thank you for your reply (and sorry for the mislabelled message).
Is there currently any public class member I can read out to find the maximum data packet size (after MTU negotiation with the BLE counterpart)?

fpistm, 2024-01-16:
Probably, but don't know when it is, you can try to enable the core debug and log messaged.

closed 2024-01-16 by fpistm