stm32duino/STM32duinoBLE #71: HCISharedMemTransportClass never available

Henrydw, 2024-07-31:
## Background

OS: Mac OS Sonoma 14.5
IDE: Arduino IDE
BOARD SETTING: Discovery
VARIANT SETTING: STM32WB5MM-DK
OPTIMISATION: None
C RUNTIME LIBRARY: Newlib Nano (default)
UPLOAD METHOD: SWD

HARDWARE: STM32WB5MM-DK Discovery board
WIRELESS STACK: stm32wb5x_BLE_HCILayer_fw.bin 
VERSION: 1.20.0
EXTRA HARDWARE: uBlox ZED F9P (I2C connection)

PACKAGES:
- STM32duinoBLE (1.2.8)
- Wire @ 2.50700.210515 (5.7.0)
- SparkFun_u-blox_GNSS_Arduino_Library (2.2.27)
- ArduinoJson (7.1.0)

## Issue Description

A function is sending data over BLE.

This data is received from an I2C connected peripheral (the uBlox GNSS chip).

Sometime on some occasions (not clear when) the BLE HCI gets stuck in an infinite loop:

The call stack looks like so:

→ sendPVTData (my function to send data received from I2C peripheral over BLE)

→Some BLE characteristic stuff……

→ HCIClass::sendAclPkt

**→ While ( _pendingPkt >= _maxPkt )**

*This while loop is never exited*

→ HCIClass::poll

**→ HCISharedMemTransportClass::available()**

**→ always returns false (`_read_index` = `_write_index`)

## Issue Reproduction

It is an infrequent runtime issue.
It always takes place within 15 minutes of turning the system on.

I will work on making an example to demo this.

## Expected Behaviour

_pendingPkt should decrease to less than  _maxPkt ?
During this time the 

This is a guess, I will expand as I dig deeper.

fpistm, 2025-03-28:
Hi @Henrydw 
probably duplicate of #58.
Could you test with new STM32duinoBLE version, please?

federica-ven, 2025-04-06:
Same thing seems to be happening to me. 

Trying to use the new version of STM32duinoBLE 1.3.9 with a Generic STM32WB55VGYx board. 

Seems like the HCISharedMemTransportClass is never available?

```
~/Documents/Arduino/libraries/STM32duinoBLE/src/utility/HCISharedMemTransport.cpp:781:39: error: 'HCISharedMemTransport' was not declared in this scope; did you mean 'HCISharedMemTransportClass'?
  781 | HCITransportInterface& HCITransport = HCISharedMemTransport;
      |                                       ^~~~~~~~~~~~~~~~~~~~~
      |                                       HCISharedMemTransportClass
```

Henrydw, 2025-04-07:
We have overcome this but it was some time ago.

I can say for certain that I have **not modified STM32duinoBLE** to over come this, we currently use **version 1.2.8**.

CAUSE: **voltage ripple** on the supply leading to stability issues on the STM32 WB BLE coprocessor.

The coprocessor would crash irreparably (requiring a reset) at irregular intervals (sometimes hours, sometimes minutes).

_NB: The tolerated voltage ripple on the STM32 WB is ~10mV._

SOLUTION:

The voltage ripple was caused by a uBlox GNSS receiver which was much noisier than we anticipated.

We solved this with a capacitance multiplier and separate LDOs but it is not a situation we are happy about.

closed 2025-04-07 by fpistm