stm32duino/STM32duinoBLE #75: Support other STM32WB target

fpistm, 2025-04-07:
> Same thing seems to be happening to me. 
> 
> Trying to use the new version of STM32duinoBLE 1.3.7 with a Generic STM32WB55VGYx board. 
> 
> Seems like the HCISharedMemTransportClass is never available?
> 
> 
> ```
> ~/Documents/Arduino/libraries/STM32duinoBLE/src/utility/HCISharedMemTransport.cpp:781:39: error: 'HCISharedMemTransport' was not declared in this scope; did you mean 'HCISharedMemTransportClass'?
>   781 | HCITransportInterface& HCITransport = HCISharedMemTransport;
>       |                                       ^~~~~~~~~~~~~~~~~~~~~
>       |                                       HCISharedMemTransportClass
> ```
> 
> it works if I revert back to an older version (ie. 1.2.9) 

 _Originally posted by @federica-ven in [#71](https://github.com/stm32duino/STM32duinoBLE/issues/71#issuecomment-2781545735)_

closed 2025-04-07 by fpistm

federica-ven, 2025-04-07:
> So, a check was done if the target is part of the supported one which is not the case of the Generic WBxxx.
> I will change this to simply warn user the library was not tested with the selected board.

Thank you Frederic, so the STM32WB55VGYx boards are not supported?

fpistm, 2025-04-07:
> Thank you [@fpistm](https://github.com/fpistm) , so the STM32WB55VGYx boards are not supported?

Generic is a basic variant to support board based on the targeted mcu. By default generic used a clock config based on internal oscillator HSI,... This library was tested for specific board with a dedicated clock config using HSE.
As the BLE for the STM32WB required to use HSE the generic cannot work by default if you do not change the clock config.
STM32WB55VGYx is know to work as the STM32WB5MM-DK is based on of this mcu. Tested using the clock config from the DK.

https://www.st.com/en/microcontrollers-microprocessors/stm32wb5mmg.html

> Description
> STM32WB5MMG is an ultra-low-power and small form factor certified 2.4 GHz wireless module. It supports Bluetooth® Low Energy 5.4, Zigbee® 3.0, OpenThread, dynamic, and static concurrent modes, and 802.15.4 proprietary protocols. Based on the STMicroelectronics STM32WB55VGY wireless microcontroller, STM32WB5MMG provides best-in-class RF performance thanks to its high receiver sensitivity and output power signal. Its low-power features enable extended battery life, small coin-cell batteries, and energy harvesting.

PR #76 (2025-04-07): chore(wb): warn user about untested board
Fixes #75
commit fb3da3bf (2025-04-07): chore(wb): warn user about untested board

Fixes #75