stm32duino/STM32duinoBLE #81: Cannot pair device

joseph-petersen-spectrumdefensesystems, 2025-07-15:
Not a bug, but I cannot find any other information anywhere on pairing / bonding.

I have set the device as pairable but am unable to pair with anything. I understand this is very little to go off of, but I am more so wondering if there is a step I am missing, like some other method I need to call when bonding? Otherwise I can provide more code.

fpistm, 2025-07-15:
Could you be more precise? Setup? Configuration? ....

joseph-petersen-spectrumdefensesystems, 2025-07-15:
Sure, here is my code:

[image]

[image]

I am able to connect when I use the nRF Connect and ST BLE Toolbox Apps, but when I attempt to bond / pair with them it fails and the connection is broken.

joseph-petersen-spectrumdefensesystems, 2025-07-16:
[stm32_ble_test.zip](https://github.com/user-attachments/files/21260043/stm32_ble_test.zip)

I have attached the entire program too if that helps.

closed 2025-07-18 by fpistm